makerbase-mks/MKS-Robin-E3-E3D #84: Bootloader

krzysziekz, 2021-02-26:
Hi.

I've E3D. How to update bootloader, the startup "beep" is very annoying. Is that safe process?

[booloaders](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/master/firmware/Bootloader)

mks-viva, 2021-03-01:
Hi! You can use Jflash download it

ll777, 2021-03-01:
the file can be copied to the sd card or it requires an additional hardware to flash it?

mks-viva, 2021-03-05:
> the file can be copied to the sd card or it requires an additional hardware to flash it?

No

mks-viva, 2021-03-05:
https://github.com/makerbase-mks/MKS-Robin-E3-E3D/issues/31

ll777, 2021-03-06:
Oh OK, thanks I was reading this post  #43 and I  have a usb to serial adapter like this one https://pt.aliexpress.com/item/1005001877595087.html Would it work?

mks-viva, 2021-03-08:
I recommend you to use JLINK

closed 2021-03-29 by mks-viva

shawe, 2022-02-13:
Can be flashed from an Arduino?