makerbase-mks/MKS-Robin-E3-E3D #85: I need firmware ".bin" file for ENDER 5 WE SETILL WAITING FOR THIS YOU TOLD AFTER 24 FEB!!!!

azatalp, 2021-03-03:
I'try to do any kind of think for setup on card. But whatever I try cant setup on mks e3 card. Every time gave me error. How 2.0.6.1 is freeze on screen after %100. Please just give working 2.0.6 fw .bin file.

mks-viva commented 23 days ago
Hi！I am sorry！We have not add bin file for ender5，we will add it after february 24

 
arsene-r2 commented 3 days ago
Hello, any updates on this?
azatalp commented 6 minutes ago
I'm still waiting for this..

mks-viva, 2021-03-06:
Hi! I will update firmware for ender5, can you test it? Because i have not ender5 machine

mks-viva, 2021-03-06:
https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/master/firmware/V1.1

mks-viva, 2021-04-01:
I will close this issues temporily

closed 2021-04-01 by mks-viva