makerbase-mks/MKS-Robin-E3-E3D #86: mks robin E3D does not update firmware.

NachoFrutos, 2021-03-08:
The board does not update the firmware through the microSD slot (TF CARD). Three different FAT32-formatted cards have been tested. Neither. The marlin version is 2.0.7. The printer is a custom one. I attach the compiled .bin file and the files configuration.h / configuration_adv.h

[marlin 2.0.7.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/6104809/marlin.2.0.7.zip)

mks-viva, 2021-03-13:
Hi! I has test your firmware on my board and it cann't update firmware. Can you provide your firmware source code to me?

NachoFrutos, 2021-03-13:
Hello, I am attaching the entire project so that you can compile it. The problem is that it updates the firmware whenever it wants. I also inform you that the BLTOUCH does not work for me as end stop z. This all well wired according to the information on the wiki. The M280 commands do not work. Tell you that if I put a mks gen L board, everything works perfectly. So I don't know if it's the marlin configuration with the BLTOUCH. But this is all according to the wiki.

SOUERCE CODE:

https://drive.google.com/file/d/1vmPtLTwbKKEtbaFXXKdVWzMjOhyO3itn/view?usp=sharing

mks-viva, 2021-03-16:
Maybe you need to format your TF card, or replace it with another TF card to test it

runobreis, 2021-04-06:
Similar issue here..The firmware updates but stops at the marlin intro logo, I've already tried all versions on this link: https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/master/firmware/V1.1/Firmware%20bin%20for%20ender3

sergiofagundes, 2021-09-17:
FIXED 100% working:

the board if using a 4GB sdcard,
I tried with 8GB/16GB/32GB and none of them worked.
So I tried a 4GB sdcard and it worked.

I used the Sdcard on the LCD sdcard slot, but I think it should work also in the mks robin board sdcard slot also with only 4GB sdcard.

The sdcard I used that worked: 4GB SanDisk Class 4 HC
The sdcars I used that didn't work: 8GB/16GB/32GB generic Class 10 HC

dtee, 2021-11-09:
I am having the same issue w/ klipper. Tried manual copy of firmware file, tried using flash-sdcard.sh - nothing seem to work, it just won't update the firmware. I am not even sure how it got 'v0.9.1-580-g478f26ca' in the first place from the same sd card (I am not sure if that's the version downloaded from mks site or something i compiled.). Here are the logs

```
# pi @ craftbot in ~/klipper on git:2a964a46 x [5:39:13]
$ ./scripts/update_mks_robin.py out/klipper.bin out/Robin_e3.bin

# pi @ craftbot in ~/klipper on git:2a964a46 x [5:39:20]
$ ./scripts/flash-sdcard.sh /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 mks-robin-e3
Flashing /home/pi/klipper/out/klipper.bin to /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
Converting Klipper binary to MKS Robin format...Done
Checking FatFS CFFI Build...
Connecting to MCU....Connected
Checking Current MCU Configuration...Done
MCU needs restart: is_config=1, is_shutdown=0
Attempting MCU Reset...Done
Waiting for device to reconnect...Done
Connecting to MCU.........................Connected
Initializing SD Card and Mounting file system...

SD Card Information:
Version: 2.0
SDHC/SDXC: True
Write Protected: False
Sectors: 7829504
manufacturer_id: 40
oem_id: BE
product_name: ND4GB
product_revision: 2.34
serial_number: 56F23A7F
manufacturing_date: 8/2017
capacity: 3.7 GiB
fs_type: FAT32
volume_label: card
volume_serial: 2900737796
Uploading Klipper Firmware to SD Card...Done
Validating Upload...Done
Firmware Upload Complete: Robin_e3.bin, Size: 21376, Checksum (SHA1): E592217D24482127D48FFBD69EDDDC0EC4E66B33
Attempting MCU Reset...Done
Waiting for device to reconnect...Done
Connecting to MCU.........................Connected
Verifying Flash...
SD Card Flash Error: Version Mismatch: Got 'v0.9.1-580-g478f26ca...', expected 'v0.10.0-0-g2a964a46...'
Traceback (most recent call last):
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 1159, in main
    spiflash.run()
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 1113, in run
    self.run_reactor_task(self.run_verify)
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 1095, in run_reactor_task
    k_reactor.run()
  File "/home/pi/klipper/klippy/reactor.py", line 269, in run
    g_next.switch()
  File "/home/pi/klipper/klippy/reactor.py", line 310, in _dispatch_loop
    timeout = self._check_timers(eventtime, busy)
  File "/home/pi/klipper/klippy/reactor.py", line 156, in _check_timers
    t.waketime = waketime = t.callback(eventtime)
  File "/home/pi/klipper/klippy/reactor.py", line 48, in invoke
    res = self.callback(eventtime)
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 1084, in run_verify
    self.new_dictionary)
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 966, in verify_flash
    json.loads(req_dictionary)['version']))
SPIFlashError: Version Mismatch: Got 'v0.9.1-580-g478f26ca...', expected 'v0.10.0-0-g2a964a46...'

```