makerbase-mks/MKS-Robin-E3-E3D #88: PWC issue

ptitpuma, 2021-03-14:
hi , i,m trying to use a pwc board on my robin E3D with tft 35but i;ve got some troubles....
when i enter the code 
  #if ENABLED(PSU_CONTROL)
    #ifndef PS_ON_PIN
       #define PS_ON_PIN         PC5    //PW_OFF, you can change it to other pin
    #endif
it show me that:
Verbose mode can be enabled via `-v, --verbose` option
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/stm32f1/pins_MKS_ROBIN_E3.h:36:0,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/pins.h:492,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:34,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:31:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/stm32f1/pins_MKS_ROBIN_E3_common.h:198:0: error: unterminated #if    
 #if ENABLED(PSU_CONTROL)
 
CalledProcessError: Command '"C:\Users\lalon\.platformio\packages\toolchain-gccarmnoneeabi\bin\arm-none-eabi-g++.exe" -DTARGET_STM32F1 -D__MARLIN_FIRMWARE__ -DARDUINO_ARCH_STM32 -DDEBUG_LEVEL=0 -DSS_TIMER=4 -D__MARLIN_DEPS__ -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-dependencies.h' returned non-zero exit status 1.:
  File "C:\Users\lalon\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 175:
    env.SConscript(item, exports="env")
  File "C:\Users\lalon\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\lalon\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\lalon\Downloads\Marlin-bugfix-2.0.x (1)\Marlin-bugfix-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 327:
    apply_features_config()
  File "C:\Users\lalon\Downloads\Marlin-bugfix-2.0.x (1)\Marlin-bugfix-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 141:
    if not env.MarlinFeatureIsEnabled(feature):
  File "C:\Users\lalon\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\lalon\Downloads\Marlin-bugfix-2.0.x (1)\Marlin-bugfix-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 292:
    load_marlin_features()
  File "C:\Users\lalon\Downloads\Marlin-bugfix-2.0.x (1)\Marlin-bugfix-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 280:
    define_list = subprocess.check_output(cmd, shell=True).splitlines()
  File "C:\Users\lalon\.platformio\python3\lib\subprocess.py", line 411:
    **kwargs).stdout
  File "C:\Users\lalon\.platformio\python3\lib\subprocess.py", line 512:
    output=stdout, stderr=stderr)

i've try with the lcd code and it compile but when i flash i've got Printer Halted
i've try to uncomment the line which are not in the first code and the printer boot but when it's time to shutdown the printer say closing machine but never do it 

what is the solution please

mks-viva, 2021-03-16:
Can you provide your firmware source code to me?

ptitpuma, 2021-03-16:
I'm using marlin bugfix 2.0.x
I can share my config.h and config adv.h tonight

ptitpuma, 2021-03-18:
[config.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/6161692/config.zip)

mks-viva, 2021-03-21:
Ok ! I will test it and use the latest marlin

ptitpuma, 2021-04-09:
Hi have test ?

mks-viva, 2021-04-22:
I has test the latest version marlin bigfix firmware and upload it for you. Marlin has change same partments and i has update our Wiki about hou to use MKS PWC module.
[Robin_e3-TEST PWC.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/6354900/Robin_e3-TEST.PWC.zip)

[Marlin-bugfix-2.0.x-TEST PWC.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/6354896/Marlin-bugfix-2.0.x-TEST.PWC.zip)

ptitpuma, 2021-04-24:
Hi is .bin configure with my marlin setting ? Because the bugfix need to be set

mks-viva, 2021-04-24:
No, you need reconfig it according to your config file. I just tested the PWC function and it works normally.

ptitpuma, 2021-04-24:
Ok I'll try 😉

ptitpuma, 2021-04-24:
I've build with your marlin bugfix but it still doesn't work when I push on button the tft still say closing machine with no effect and the LCD show me killed printer halted please reset

ptitpuma, 2021-04-26:
I've found something the code you tell was wrong I must write this 
#if ENABLED(PSU_CONTROL)
  #ifndef PS_ON_PIN
    #define PS_ON_PIN         PC5    //PW_OFF, you can change it to other pin
  #endif  
#endif

But when I push stop button the tft still say closing machine and the LCD show me power off but still stuck on this with no effect and I've found a bug the tft35 isn't working well when I send a print it start heating the bed but on the tft no temp is moving and when the bed is at the right temp the hotend never start heating it's like no order is send and no info are received from the board to the tft

mks-viva, 2021-04-27:
I don’t know you used MKS TFT screen, then you should refer to it WIKI (With the MKS TFT)：https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/MKS_PWC
You need remove it in pins_MKS_ROBIN_E3_common.h file
```
  #ifndef KILL_PIN
    #define KILL_PIN          PB10    //PW_DET, you can change it to other pin
  #endif
  #define KILL_PIN_INVERTING  true     //true : HIGH level trigger
  #define KILL_PIN_STATE      KILL_PIN_INVERTING  
```
[image]

mks-viva, 2021-04-27:
PW_OFF signal connect to E3's PC5
PW_DET signal connect to TFT's PB4

ptitpuma, 2021-04-27:
> I don’t know you used MKS TFT screen, then you should refer to it WIKI (With the MKS TFT)：https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/MKS_PWC
> You need remove it in [...]

I've done this

ptitpuma, 2021-04-27:
> PW_OFF signal connect to E3's PC5
> PW_DET signal connect to TFT's PB4

So I think here is the issue about TFT not working as I have the LCD connect on E3 exp 1 the PC5 is already use how can I do of I want my TFT and my LCD ?

mks-viva, 2021-04-30:
Can you provide your board wiring to me? May be i need check it.

ptitpuma, 2021-04-30:
Ok I have found a part off my trouble it was a faulty wire... I have on my board plug TFT on aux 1 
Plug PW det on TFT pb4 and plug PW Off to PC5 on the board (this wire was faulty) now I still have trouble ( but the pcw work ) when I launch a print on TFT it never start, when it's by the E3 board SD card no trouble I have reflash the tft no change...  And I need to know if it's normal that now my printer stop alone

ptitpuma, 2021-05-18:
@mks-viva

mks-viva, 2021-05-19:
Was it possible to print normally before? Can the temperature still be detected?

ptitpuma, 2021-05-19:
Yes it was possible before... Now when I launch a print no temperature are detected

mks-viva, 2021-05-19:
You need check you temperature wiring

ptitpuma, 2021-05-19:
If I launch a print on my LCD every temperature are good and working if I do preheat on the TFT everything works well it's only when I launch a print that trouble arrive like in this post 
https://github.com/makerbase-mks/MKS-TFT/issues/225

ptitpuma, 2021-05-30:
@mks-viva

bevanweiss, 2021-09-18:
So you have BOTH a TFT LCD, and another LCD connected?
What are your connections for this?

Your config file doesn't mention that you have the TFT35
//
// 480x320, 3.5", FSMC Display From MKS
// Normally used in MKS Robin Nano V1.2
//
//#define MKS_ROBIN_TFT35

And if you have an LCD which is using LCD_D7 (PC5), then using PC5 for your PWC would not work.  You would need to select a different IO pin which is NOT already in use by another of your interfaces (i.e. not LCD, not TFT, not LIMITS / PROBE / NEOPIXEL / STEPPERS etc)

ptitpuma, 2021-10-01:
Hi @bevanweiss 
I have my original cr10 LCD and my TFT connected my LCD is connected on  ender3/exp1 and my TFT on aux-1... 
Should I mention TFT on marlin as it was working well before I try to make pwc working ? 
Which IO pin should I select ? I haven't got NEOPIXEL on my printer can I use one of this ? Thanks for your help btw

bevanweiss, 2021-10-02:
Yeah, the NeoPixel data pin (PA2) should work.

ptitpuma, 2022-06-13:
@bevanweiss 
I need your help today I have tried to update my printer but when I try to comment my TFT on marlin I always get errors could you help me please