makerbase-mks/MKS-Robin-E3-E3D #91: how to use chinese simplified in ender3 default display and E3 motherboard

zyystudio, 2021-03-25:
I have gone through the site and successfully compiled the 2.0.6.1 marlin with E3 and 3d touch with my parameters to flash the ender3.
However I do see there is simplified chinese version for E3D. 
Is there documentation how to do that?
Thanks
Stanley

zyystudio, 2021-03-26:
BTW the display model on ender3 by default is LCD12864

zyystudio, 2021-03-26:
#define LCD_LANGUAGE zh_CN

zyystudio, 2021-03-26:
?
It seems overflow the 256K of STM32F103RCT6 right?

mks-viva, 2021-03-26:
Yes, if you use Chinese, it is easy to cause the Flash to be larger than 256kb. Therefore, you need to block the boot logo, etc. Maybe, you refer to **Marlin-2.0.6.1_E3D_V1.0_Chinese_TMC2209** source code and it's will make sense.

zyystudio, 2021-03-26:
aha! it's you again! Thanks. That's a good idea! Lovely!

zyystudio, 2021-03-26:
Wonderful! Solved! shrink several ones! Love you!

mks-viva, 2021-03-27:
Haha! So, i will close this issues temporily.

closed 2021-03-27 by mks-viva