makerbase-mks/MKS-Robin-E3-E3D #92: Cant connect to Printer via USB

Sebe47, 2021-03-25:
Hello im using the "Marlin-2.0.6.1_for_ender3_TMC2209+3dtouch+Z_SAFE_HOMING" Firmware 

I cant get managed to connect to the Printer via USB 

Tryed it with my Computer and with Octoprint.

mks-viva, 2021-03-26:
Hi! If you use pronterface connect to board and it's work ok?

aabeuge, 2021-04-29:
FWIW, I believe the CH340 usb to serial chips (which Robin E3 uses) can be damaged if the raspberry pi is connected to the board without its own external power.  This is because the pi 5v rail is connected to its usb connector so the pi will try to power itself through the usb connection to the Robin E3 board.  When this happens too much current moves through the ch340 chip and kills it.  I discovered this while using ramps boards with a mega2560 clone that had the ch340 chip on it and killed three boards within a couple days.  I bought some replacement ch340 chips off eBay for cheap and replaced mine and all three boards usb worked again after.

You can try to use the AUX1 serial connection (refer to the hardware section of the git for pin out) with separate usb FTDI or connect raspberry pi through it's serial pins.  If it works then your ch340 chip might need replaced or just continue  to use the AUX1 serial connection permanently.

countcobolt, 2021-07-01:
Make sure you have enabled the second serial port if you are using UART. The first one is already in use.

closed 2021-08-03 by mks-viva