makerbase-mks/MKS-Robin-E3-E3D #93: MKS robin E3D use MKS OLED no working

a4870010, 2021-03-26:
Use MKS robin e3d motherboard, connect MKS OLED0.96 display, but no working and i has add pin define to pins_MKS_ROBIN_E3_common file
```
  #if ENABLED(MKS_12864OLED_SSD1306)
        #define LCD_PINS_DC          PA7  // Set as output on init
        #define LCD_PINS_RS          PC4  // Pull low for 1s to init
        #define DOGLCD_A0            LCD_PINS_DC
        #define DOGLCD_CS            PA5
        #define DOGLCD_MOSI          PA4
        #define DOGLCD_SCK           PA6
  #endif
```
[Config.and.pin.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/6211656/Config.and.pin.zip)

Use RAMPS1.4 motherboard and MKS OLED0.96 will working
I am run offical marlin bug-fix 2.0.x version

mks-viva, 2021-04-07:
May be i need test it

mks-viva, 2021-08-10:
Marlin have not support SPI oled on STM32, so, i suggest you to consider using MKS MINI12864 V3. I will close this issues.

closed 2021-08-10 by mks-viva