makerbase-mks/MKS-Robin-E3-E3D #95: error: #error "No E stepper plug left for Z2!

Niptoc, 2021-03-30:
Hello,

I want to compile MKS Robin E3 V1.0 for a CR-10 with a dual Z motor with marlin  2.0.7.2 and I get an error when compiliing:

``buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/pins.h:1410:8: error: #error "No E stepper plug left for Z2!``

mks-viva, 2021-04-01:
Hi! MKS Robin E3 have not E1 axis, you can use Z2 interface

Niptoc, 2021-04-01:
Hi, Sorry but I'm not sure to understand. 
What do you mean by using Z2 interface ?

mks-viva, 2021-04-01:
[image]
Z1 and Z2 interface

Niptoc, 2021-04-01:
This is what I plan to use. But are the Z1 and Z2 independant (Can we use them independantly) so that I can use the implemented feature to autocalibrate the Z axis.
When I check the drawing for PINs and the PIN configuration file there are no distinction for Z1 and Z2.

mks-viva, 2021-04-02:
They are connected in parallel. Since MKS Robin E3 has only 4 axes, you cannot define independent Z2

Niptoc, 2021-04-02:
Ok, thank you for the answer, it is clear now :)

closed 2021-04-02 by Niptoc