makerbase-mks/MKS-Robin-E3-E3D #96: Compilation error

Niptoc, 2021-03-30:
Hello

I also have compilation error for a CR10.
I want to use the TFT 35 and I got the error message :
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0, from Marlin\src\HAL\STM32F1\HAL.cpp:30: Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:839:6: error: #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)." #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."

I have activated the ADVANCED_PAUSE_FEATURE to be able to use the filament detector.

Do anyone encounter the same issue?

mks-viva, 2021-04-01:
Hi! If you want to use filament detector function, you can MKS TFT35's filament detector function and does't setting marlin.

Niptoc, 2021-04-01:
Ok, I removed from Marlin the filament runout detector and deactivated the ADVANCED_PAUSE_FEATURE.

I am using latest Marlin version 2.0.7.2 and I got new error messages.
I am compiling using the `env: mks_robin_e3` . I hope this is the right one.

```
In file included from Marlin\src\HAL\STM32F1\tft\tft_fsmc.cpp:28:0:
C:\Users\user\.platformio\packages\framework-arduinoststm32-maple\STM32F1\system\libmaple\include/libmaple/fsmc.h:47:2: error: #error "FSMC is unavailable on your MCU"
 #error "FSMC is unavailable on your MCU"
  ^~~~~
Marlin\src\HAL\STM32F1\tft\tft_fsmc.cpp: In static member function 'static void TFT_FSMC::Init()':
Marlin\src\HAL\STM32F1\tft\tft_fsmc.cpp:89:16: error: 'FSMC_CS_PIN' was not declared in this scope
   uint8_t cs = FSMC_CS_PIN, rs = FSMC_RS_PIN;
                ^~~~~~~~~~~
Marlin\src\HAL\STM32F1\tft\tft_fsmc.cpp:89:16: note: suggested alternative: 'FSMC_CS_NE1'
   uint8_t cs = FSMC_CS_PIN, rs = FSMC_RS_PIN;
                ^~~~~~~~~~~
                FSMC_CS_NE1
Marlin\src\HAL\STM32F1\tft\tft_fsmc.cpp:115:11: error: 'rs' was not declared in this scope
   switch (rs) {
           ^~
Marlin\src\HAL\STM32F1\tft\tft_fsmc.cpp:115:11: note: suggested alternative: 'cs'
   switch (rs) {
           ^~
           cs
Marlin\src\HAL\STM32F1\tft\tft_fsmc.cpp:155:17: error: 'GPIOE' was not declared in this scope
   gpio_set_mode(GPIOE,  7, GPIO_AF_OUTPUT_PP);  // FSMC_D04
                 ^~~~~
Marlin\src\HAL\STM32F1\tft\tft_fsmc.cpp:155:17: note: suggested alternative: 'GPIOD'
   gpio_set_mode(GPIOE,  7, GPIO_AF_OUTPUT_PP);  // FSMC_D04
                 ^~~~~
                 GPIOD
Marlin\src\HAL\STM32F1\tft\tft_fsmc.cpp:172:25: error: 'rs' was not declared in this scope
   gpio_set_mode(PIN_MAP[rs].gpio_device, PIN_MAP[rs].gpio_bit, GPIO_AF_OUTPUT_PP);  //FSMC_RS_Ax
                         ^~
Marlin\src\HAL\STM32F1\tft\tft_fsmc.cpp:172:25: note: suggested alternative: 'cs'
   gpio_set_mode(PIN_MAP[rs].gpio_device, PIN_MAP[rs].gpio_bit, GPIO_AF_OUTPUT_PP);  //FSMC_RS_Ax
                         ^~
                         cs
*** [.pio\build\mks_robin_e3\src\src\HAL\STM32F1\tft\tft_fsmc.cpp.o] Error 1
```

Do you know how I can solve this?

mks-viva, 2021-04-01:
Can you provide your config to me?

Niptoc, 2021-04-01:
Here are the config (renamed in .txt)
[pins_MKS_ROBIN_E3.txt](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/6243251/pins_MKS_ROBIN_E3.txt)
[pins_MKS_ROBIN_E3_common.txt](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/6243252/pins_MKS_ROBIN_E3_common.txt)
[Configuration.txt](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/6243242/Configuration.txt)
[Configuration_adv.txt](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/6243243/Configuration_adv.txt)

In case this is not enough here is whole Marlin file. 
Linked to this issue I have also the double Z axis. But to try to reach a first compilable file I have set to 1 Z axis, BLTouch enable, No runout filament detector (but I have one so I would like to make it work also)
[Marlin-2.0.7.2.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/6243261/Marlin-2.0.7.2.zip)

mks-viva, 2021-04-02:
You need set it in configuration.h file
1. Set : #define SERIAL_PORT 1 // For MKS TFT35
2. Disable "#define MKS_ROBIN_TFT35", this is not MKS TFT35
You can refer to Wiki about 'How to use MKS TFT': https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/Display_MKS_TFT24_35_70

Niptoc, 2021-04-02:
Thank you for your support ! 
I didn't see this link  for the screen

closed 2021-04-02 by Niptoc