makerbase-mks/MKS-Robin-E3-E3D #97: Sensorless homing with z_axis never work

pinball83, 2021-04-03:
- Setup jumpers zdiag
- Turn of x/y/z endstop
- Enable Sensorless homing for all axis x/y/z
- Configuration_adv.h: 

```
  #define SENSORLESS_HOMING // StallGuard capable drivers only

  #if EITHER(SENSORLESS_HOMING, SENSORLESS_PROBING)
    // TMC2209: 0...255. TMC2130: -64...63
    #define X_STALL_SENSITIVITY  200
    #define X2_STALL_SENSITIVITY X_STALL_SENSITIVITY
    #define Y_STALL_SENSITIVITY  200
    #define Y2_STALL_SENSITIVITY Y_STALL_SENSITIVITY
    #define Z_STALL_SENSITIVITY  200
    #define Z2_STALL_SENSITIVITY Z_STALL_SENSITIVITY
```

- Configuration.h
 
```
#define INVERT_X_DIR false
#define INVERT_Y_DIR false
#define INVERT_Z_DIR true
```

```
// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING false // Set to true to invert the logic of the probe.
```

X and Y sensorsless homing working correct
Z did not

mks-viva, 2021-04-07:
I suggest you don’t use this function for Z axis

mks-viva, 2021-04-24:
I will close this issues temporarily

closed 2021-04-24 by mks-viva