makerbase-mks/MKS-Robin-E3-E3D #98: Unable to flash self-compiled binary

Stalkk, 2021-04-06:
Hello,

i have a issue with E3 V1.1 and Ender 3. I am able to compile the binary with VisualStudio code and platform.io but flashing from SD  (yes i renamed the binary  to Robin_e3.bin) **is stopped on 100%**. After reboot it hangs in **"boot"** screen. 

The precompiled Marlin-2.0.6.1_for_ender3_TMC2209.bin can be flashed and works but i need to change the config for my printer and compile the own version. 

The other point i noticed is - the compiled firmware.bin file is only 185 kb big and the precompiled binary (Marlin-2.0.6.1_for_ender3_TMC2209.bin) is about 217 kb. 

Thank you for your help.

mks-viva, 2021-04-07:
Hi! You need copy Robin_e3.bin file to TF card.
Doesn't use firmware.bin rename as Robin_e3.bin

Stalkk, 2021-04-07:
Hi,

thank you for your answer.  As i already wrote (yes i renamed the binary to Robin_e3.bin)  i know how to flash the firmware and it works with your precompiled binary. But here are the steps i did:

1. Compile the Marlin  from https://github.com/makerbase-mks/MKS-Robin-E3-E3D
2. rename firmware.bin to Robin_e3.bin
3. copy Robin_e3.bin to sd card
4. install the sd card into the board an turn it on
5. flashing process started and stops by 100%
6. after reboot is stops on "boot" screen, Robin_e3.bin is renamed into Robin_e3.cur 

Pleae read again my description

mks-viva, 2021-04-07:
You need copy Robin_e3.bin and cann't use firmware.bin rename as Robin_e3.bin

Stalkk, 2021-04-07:
Hello mks-viva, 
You're right. 
I just missed the file. Ashes all over my head. 
It works fine now with the right file. 
Thank you!

closed 2021-04-07 by Stalkk