makerbase-mks/MKS-Robin-E3-E3D #101: Firmware update stopped on "update 100%"

piontec, 2021-04-17:
I configured the latest marlin-bugfix2.0.x according to your instructions. My board is v1.1. The firmware compiled and started to flash successfully, but then it's stuck on "update 100%". After removing the card, the printer won't boot and is stuck with message "boot" on the screen. THe firmware file is deleted from the sdcard.

mks-viva, 2021-04-19:
Hi! You need copy Robin_e3.bin file to TF card and insert board update it

piontec, 2021-04-19:
Hey! Thank you, I missed that. What is the difference between `firmware.bin` and `Robin_e3.bin`? I'm trying to make your board work with https://github.com/KevinOConnor/klipper and I have the same problem there - I guess the solution is the same. Can you help me convert klipper's binary as well?

mks-viva, 2021-04-20:
Robin_e3.bin file is obtained by transcoding using firmware.bin file, so, we need use Robin_e3.bin file update firmware and cann't use firmware.bin rename as Robin_e3.bin, this is unusable.

piontec, 2021-04-20:
Yes, but how to transcode the file?

mks-viva, 2021-04-20:
You can refer to marlin about transcode and i has upload this file.
[mks_robin_e3.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/6341338/mks_robin_e3.zip)
[image]

mks-viva, 2021-04-20:
You can refer to Klipper here about how to transcode: https://github.com/KevinOConnor/klipper/blob/master/scripts/update_mks_robin.py

piontec, 2021-04-20:
Thanks for the info, but exactly this script doesn't work for me - when I flash `Robin_e3.bin` from it, the printer hangs like with non-encoded firmware :( Just to confirm: which bootloader size and serial ports should I use for klipper?

spooky79, 2021-04-21:
I have the same problem, it comes to 100% and nothing is happening, I'm using visual studio code, any solution?

mks-viva, 2021-04-22:
@piontec About MKS Robin E3 use kilpper firmware settigs: https://github.com/KevinOConnor/klipper/pull/3008
@spooky79 What firmware you using?

piontec, 2021-04-26:
It doesn't work for me, see my marlin report here: https://github.com/KevinOConnor/klipper/issues/4219

piontec, 2021-04-29:
It turned out to be klipper issue, it was silently crashing without any info in logs. Closing as invalid.

closed 2021-04-29 by piontec

thiagoedson, 2021-09-22:
= aqui

fex42, 2021-11-03:
I also have the same problem with Klipper. My workaround was to get back to the tagged v0.10.0 version and flash this. After flashing the board, you can go back to git HEAD for the stuff running on the Raspberry Pi

bulcke, 2022-09-10:
> I also have the same problem with Klipper. My workaround was to get back to the tagged v0.10.0 version and flash this. After flashing the board, you can go back to git HEAD for the Klipper stuff running on the Raspberry Pi

@fex42 where can I find said tagged version?