makerbase-mks/MKS-Robin-E3-E3D #102: Error compiling Marlin bugfix-2.0.x when serial port is set to-1

daBigR--, 2021-04-17:
Hi, I downloaded Marlin bugfix-2.0.x from official site.  I used example Configuration.h and Configuration_adv.h and set up parameters for my Ender-3, BLTouch, etc.  I configured #define SERIAL_PORT -1 as indicated on this site's Wiki.  I also set #define MOTHERBOARD BOARD_MKS_ROBIN_E3D_V1_1.  When I compiled I got error  "SERIAL_PORT is set to -1, but the MOTHERBOARD has no native USB support" .  As it turned out this error is generated by this directive on the board definition in standard Marlin #define BOARD_NO_NATIVE_USB.  So in new Marlin you cannot set serial port to -1 for Robin E3 boards.  What should I do to compile Marlin and to have the USB communications working correctly?

mks-viva, 2021-04-19:
You need set SERIAL_PORT 1 when you use the latest version marlin firmware.

daBigR--, 2021-04-19:
OK.  And SERIAL_PORT_2 ?  Should it be commented?

mks-viva, 2021-04-19:
You need disable SERIAL_PORT_2. I will update our Wiki.

mks-viva, 2021-04-24:
I will close this issues temporarily

closed 2021-04-24 by mks-viva

kacharros, 2021-06-10:
Mismo problema con marlín 2.0.8.2. 
Err0r con serial port -1.
Cambio a serial port 1 y baudrate 250000 y compila pero la tft35 no se comunica con la impresora.