makerbase-mks/MKS-Robin-E3-E3D #104: Endstopes not work

rogeriollorente, 2021-04-19:
Good afternoon, I bought an MKS Robin e3d card, when installing the card I identified that the endstopes do not work.

Initially I thought it was a problem with the wiring or the micro switches, but I tested them all and they work well.

When I send the command Gcode M119 I get the answer that the micro switches are open, even when I activate them manually.

I don't know how to solve this, can someone help me?

mks-viva, 2021-04-20:
Hi! Can you provide a video when your printer go to home?

rogeriollorente, 2021-04-21:
Hello, unfortunately I was unable to produce a video, but I took some pictures that can help you understand.

I am using a simple micro switch that I tested before and after connected to MKS E3D.

The MKS E3D is no longer connected to my printer, so I can't show you the printer going Home .... because when I tried, she didn't turn off the engines when she reached the endstop ... it was tense !!!

I'm using a TFT to send the Gcode M119 to check the status of the endstop micro switches.
[image]
[image]
[image]
[image]
[image]
[image]

I always have the same answer that they are open.

mks-viva, 2021-04-23:
You need remove X Y Z E DIAG signal connect jumpers
[image]

rogeriollorente, 2021-04-24:
For now the problem of endstopes has been solved, later I implement the sensorless homing function.

But now I have another question, which of the pins can I use as PS_ON_PIN, because I have a printer shutdown system and I would like to use it.
Thanks.

mks-viva, 2021-04-26:
You can refer to Wiki: https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/MKS_PWC

rogeriollorente, 2021-05-06:
Hello, successfully implemented the power off, added to the file pins_MKS_ROBIN_E3_common.h the line

//
// PS-ON-PIN
//
#define PS_ON_PIN PB10

But there is a new problem, I'm having a lot of vibration in stepper motors.

I am using the TMC2209, with the default settings of the official Marlin for tmc, considering that they are more or less similar to the ones I obtained through calculations of Vref = 730ma and thus Vref = 800ma.

My stepper motors are Chinese, Hanpose, 1.7a, 1.6 Ohm, model 17hs4401.