makerbase-mks/MKS-Robin-E3-E3D #105: How many LEDs can I connect to the neopixel port?

piontec, 2021-04-20:
Hey!
I have Robin E3D 1.1 and some neopixel LEDs, but I can't find any spec that says how many LEDs can I connect to the board directly? What is the max power output of the connector?

mks-viva, 2021-04-30:
Hi! MKS has reserve a 5V interface, DC-DC use MP584(Support 3A output), but has fues(limit 1.5A), so, you need limit it in 1.5A

DocYester, 2021-05-03:
Calculate 50mA per LED as max current - so 30 LED's should be the maximum to be safe

piontec, 2021-05-03:
Thank you for your responses. Just to make sure: 1.5A is when I'm connecting directly to the mainboard, without any external modules, voltage regulators and so on, right?

DocYester, 2021-05-03:
> Thank you for your responses. Just to make sure: 1.5A is when I'm connecting directly to the mainboard, without any external modules, voltage regulators and so on, right?

Yes.
But you can connect the "driving Pin" to the LED-Stripe and the power source from an external as well - to get more than 30 LED's up - dont forget to connect gnd's as well.

piontec, 2021-05-03:
Thank you, all clear now! It would be super helpful to add this to docs :)

closed 2021-08-09 by mks-viva