makerbase-mks/MKS-Robin-E3-E3D #106: BL TOUCH DOES NOT STOW WHILE TOUCHING THE BED

mustafadh12, 2021-04-27:
Hello,

I was using ender 3 pro with bl touch installed perfectly until i've shortcutted the thermistor circuit than renewed the motherboard as the same.

Triple checked the wiring, seems everyhing stays right position. Here is the problem, bl touch is working fine manually. I can control it on the menu, seems working good at the startup also. When i press level bed, it reaches the bed and wont stow/retract. Nozzle reaches the bed and i stop the printer by hand.

I've tried to change bed height manually, restored defaults on the menu and etc. 

Funny thing is, when i put a plastic part on the bed (thickness 2 mm), than command level bed, probe detects it and retracts! So I'm started to think if it's because of the hardness of the glass maybe? But older motherboard with the same bl touch was working perfectly. 

Please help me ASAP.

mustafadh12, 2021-04-27:
Here is the config files. BTW I am using both z switch and bl touch.

[Marlin.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/6382705/Marlin.zip)

mustafadh12, 2021-04-27:
Is there a possibility to damaged while thermistor circuit shortcut happened?

aabeuge, 2021-04-29:
I had a similar problem and I had to reverse the polarity of the bltouch end stop wires when connecting to z_endstop.  The white wire I believe needs to be on the right side (toward front).

This also gave another strange behavior when the wires were reversed.  When giving m119 command to see endstop trigger status the bltouch pin would drop the moment I enter the command.  And would always say triggered in the status.

countcobolt, 2021-07-01:
Does the power on self test work?