makerbase-mks/MKS-Robin-E3-E3D #107: Boot loader update

aabeuge, 2021-04-28:


mks-viva, 2021-04-30:
Hi! You can use ST-Link.

aabeuge, 2021-04-30:
> Hi! You can use ST-Link.

I tried connecting to the J1 header and the ST link utility could not find the target.  Does Boot0 have to be pulled high even for SWD programming?  If so I saw that I will need to cut the trace line from permanent ground.

mks-viva, 2021-04-30:
Like this: https://github.com/makerbase-mks/MKS-Robin-E3-E3D/issues/43
But i recommend you to use J-LINK and use it have not change hardware.

aabeuge, 2021-04-30:
Success!
I took another look at stm32cubeprogrammer and the default memory address needed to change from 0x80000000 to 0x00000000.  Once I made the change it connected using the ST-Link V2 dongle and I was able to save the original bootloader and upload the new bootloader without the beep and boot message.