makerbase-mks/MKS-Robin-E3-E3D #108: How to connect 12V filament cooling fan via DC-DC converter?

piontec, 2021-04-29:
I wanted to connect 12V fan for cooling the filament. I used a 12V Dc-DC converted in the same way as recommended for creality 4.2.2 board: I left the ground pin of filament cooling fan untouched, then inserted the DC-DC converted into +24V line and attached converter's ground directly to power supply gnd. Unfortunately, this resulted in my 12V fan still getting volatages much higher than 12V. How should I connect that?

mks-viva, 2021-04-30:
Hi! Can you provide your dc-dc module pictures or datasheet?

piontec, 2021-05-03:
Please check photos of the converter I'm using
[image]
[image]

aabeuge, 2021-05-04:
I would start by suggesting that it is best to always verify your output voltage is correct on a dc-dc converter before powering anything from it.

I think the enable pad can be left alone but I'm not 100%.  I searched the internet and found a QS-1205CME-3A module that looks like those photos and the description of the enable pin suggests it is logic controlled with 0V for off and 2-5V for on.  If not connected it likely has a pull-up resistor on the board to keep it enabled.  I'm not sure why more voltage would have gone to the fan unless your connections were wrong.

Based off of the photos I would suggest:

GND should connect to -V of your printer power supply and black/ground/negative lead of fan.

IN+ should connect to +V of printer power supply.

VO+ should be connected to red positive lead of fan but verify the voltage first and adjust for 12V.

Hope this helps

piontec, 2021-05-05:
I have the DCDC connected correctly, exactly as you described. And I measured the voltage to be 12V. Yet somehow when board switching kicks in, the output voltage is higher - I'm not really sure how's that possible...

mks-viva, 2021-05-19:
@piontec The manufacturer may have relevant tutorials, you can consult.

closed 2021-08-03 by mks-viva