makerbase-mks/MKS-Robin-E3-E3D #112: MKS-Robin-E3P Bltouch will not deploy

nfruge56, 2021-05-17:
I'm running an Ender 3 V2 and I'm replacing the motherboard. I've followed all of the videos and how-to's but I can not get the Bltouch to deploy when homing the Z-axis. Attached is the config file I'm using. 

The Bltouch will start-up and deploy with it's self-test but not when homing the z-axis.

I've tried both using the Servo method and the z-axis end stop method. Neither are working. 

Please advise.

[Configuration.txt](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/6496889/Configuration.txt)

nfruge56, 2021-05-17:
Moved to MKS-Robin-E3P

closed 2021-05-17 by nfruge56

bevanweiss, 2021-09-18:
This may have been because you didn't have the following setting configured in the file

// Force the use of the probe for Z-axis homing
#define USE_PROBE_FOR_Z_HOMING

If you don't have this in my experience it still uses the Z-LIMIT switch (if you have the Z PROBE on a separate pin to the Z LIMIT, as I do).