makerbase-mks/MKS-Robin-E3-E3D #113: CR-10 Marlin firmware - Robin E3 V1.1

jimbob111, 2021-05-19:
Is there a Marlin for the CR-10? I can only see Ender 3 & Ender 5 in the firmware folder?

countcobolt, 2021-07-01:
You need to compile it yourself. I did it using Platformio. Works fine but requires you to edit config files.

jimbob111, 2021-07-04:
Thanks, have done that and am now getting a "TMC connection error" on the x-axis

countcobolt, 2021-07-05:
Check how you address your stepper motors and make sure the jumper are in the correct spot. E.g. is you set them to use UART make sure that the jumpers are on UART. I assume you selected 1 mode and place them on another mode in the firmware. I uploaded my configuration. **Attention** This originally is when this board was installed in a CR10S5 so the bed size is bigger. Modify this size to 310*310*400. It also had a BL Touch on there. I do not recall if this firmware has PID for bed enabled or not as this comes from a back up. (I don't have access to my main machine now)

[configuration.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/6762393/configuration.zip)

jimbob111, 2021-07-05:
Thanks for the config files, I haven't tried them yet, will do later. I  have the E3 board, so don't have an option to set jumpers.

closed 2021-07-29 by jimbob111