makerbase-mks/MKS-Robin-E3-E3D #114: BTT mini 12864 v1 in E3D V1.1

kacharros, 2021-06-04:
Does anyone know instructions to install a mini BTT 12864 v1 on Robin E3D v1.1?
I uncomment FYSECT_MINI_12864_2_1 y NEOPIXEL_LED , it as the BTT instructions put it and it gives an error.

mks-viva, 2021-06-16:
Hi! You need add it in ini file
custom_marlin.NEOPIXEL_LED = Adafruit NeoPixel=https://github.com/ellensp/Adafruit_NeoPixel
[image]

mks-viva, 2021-07-15:
You can refer to https://github.com/MarlinFirmware/Marlin/pull/22368

closed 2021-07-15 by mks-viva

zhongxb, 2021-07-24:
> Hi! You need add it in ini file
> custom_marlin.NEOPIXEL_LED = Adafruit NeoPixel=https://github.com/ellensp/Adafruit_NeoPixel
> [image]

this address is 404,  https://github.com/ellensp/Adafruit_NeoPixel.