makerbase-mks/MKS-Robin-E3-E3D #115: MKSTOOL onyl mksrobin e3d 1.0version,Update to 1.1?

Verkol, 2021-06-04:
I tried to install CR20 pro on the MKS E3D version 1.1 motherboard, but the Z axis and BL touch did not work properly.
https://baizhongyun.cn/home/mkstoolview

There is probably version 1.0 above on the MKSTOOL website.Is it possible to update with 1.1version files?

If I want to install published files, the screen will fall apart, freeze or disappear. Z axis, BL touch looks good.MINI12864 display.

pepegragera, 2021-06-16:
Some problem...

Verkol, 2021-06-18:
After several days of tinkering I managed to put together a working 3d printer. On the one hand, the problem was that I didn't really look into the fact that the 1.1V motherboard had several positive changes, which needed careful attention when setting up the software (e.g. onboard eeprom). BL touch also works fine at the moment,because I had some problems with it,because M999 error kept throwing up. 
A few more days of testing would not hurt,because I have not dared to leave the machine unattended,but I do not dare to do so at the moment.

Translated with www.DeepL.com/Translator (free version)

mks-viva, 2021-07-15:
You need define mother board is MKS_ROBIN_E3D_V1_1

closed 2021-07-15 by mks-viva