makerbase-mks/MKS-Robin-E3-E3D #116: Ender 2

dexter2111, 2021-06-05:
Hello,

This board support mini 12864 lcd , that is used in Ender 2?
Can some body help me to configure the firmware for Ender 2 ?
THX.

Verkol, 2021-06-15:
Yes, it works with it. It works with the type I have.
You have to download VS studio to edit the configuration.h file.It's basically like commenting out one display and the other one the other way round.Simple text editing.

VS install:
https://www.youtube.com/watch?v=eq_ygvHF29I&t=303s
This is the part you need to comment on:
// MKS MINI12864 with graphic controller and SD support
// https://reprap.org/wiki/MKS_MINI_12864
#define MKS_MINI_12864
#define LCD_CONTRAST_MIN  140
#define LCD_CONTRAST_INIT 155
#define LCD_CONTRAST_MAX  165

Also, you need to change the 3D printer's hardware configuration information.
https://baizhongyun.cn/home/mkstoolview

This interface would help, but unfortunately it doesn't work for me anymore.
Here you can select the type that you might have.

dexter2111, 2021-06-19:
Thank you, for your replay.
It seems that the MKS tool, is not working any  more.
I have made the configure file, put it in VS , bulid the firmware, but the display is not show any thing( only the backlight is on).
The firmware has flash in to the board because the extension of the file has changed form .bin to .cur.
My display has 1 ribbon cable with 10 pin, and i connected to exp1. 
Thank you.

Verkol, 2021-07-02:
I do not have the same display,mine is connected to exp1,exp2 with two cables.
You would have to look in the Marlin firmware to see which display is commented out. You can find out exactly which display it is in the Marlin example configuration file.

dexter2111, 2021-07-15:
Thank you, for your support.
I have managed to make it working.
I have rewiring  the display signals with dupont wire.
[image]
[image]
All is working fine with:
#define MKS_MINI_12864
#define LCD_CONTRAST_MIN 140
#define LCD_CONTRAST_INIT 155
#define LCD_CONTRAST_MAX 165
Without  define the LCd Contrast , by default the contrast is 195.

mks-viva, 2021-08-09:
If you use MKS Robin E3 board, you can connect to MKS MINI12864 by EXP1 and EXP2