makerbase-mks/MKS-Robin-E3-E3D #117: MKS Robin E3D v1.1 Cant connect via USB or Aux1

NZOberoN, 2021-06-07:
Hi, very much a newbie here.

I recently bought a Robin E3D with TMC2209 drivers, connected to the board via USB and Pronterface, but needed to make a few changes, invert stops etc.

I compiled the latest Marlin and bugfix with changes and flashed firmware via SD card. Couldn't connect using Pronterface. No error, just persistant "connecting..." message.

Tried to download "factory" firmware via this page https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/Marlin_firmware but found the link doesn't work (404 error), so followed my nose to here https://github.com/makerbase-mks/MKS-Robin-E3-E3D/blob/master/firmware/V1.1/Firmware%20bin%20for%20ender3/Marlin-2.0.6.1_for_ender3_TMC2209.bin. Flashed that, still can't connect.

I also tried the full Marlin download (https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/master/firmware/V1.1/Marlin-2.0.6.1_for_ender3_TMC2209), compiled and flashed that... same problem.

I read that the CH340 chip can sometimes be the problem so I tried to connect via Aux1 with a FTDI... same Problem.

I don't have a usable screen to attach to the board, so I have no feedback whatsoever.

Thank you in advance!

aabeuge, 2021-06-09:
Silly question but did you change the firmware file name to Robin_e3.bin before flashing?  If not that would definitely be the reason.

NZOberoN, 2021-06-10:
Yes I did, but, as it turns out I needed to reflash the bootloader, it had become corrupt somehow.

closed 2021-06-10 by NZOberoN

aabeuge, 2021-06-14:
Ahh, glad you were able to sort it out.