makerbase-mks/MKS-Robin-E3-E3D #121: Robin E3D with EXP-MOT3

countcobolt, 2021-07-01:
Hi all, is it possible to use the EXP-MOT3 on a Robin E3D? I use EXP3 for my standard Ender type screen, so exp 1 & 2 are open. What files need to be modified in marlin? Using 2.0.6.1 from this branch. I would like to use this to
1. Add an additional extruder 
2. Separate my Z-steppers so I can use G34 for levelling.

kind regards

mks-viva, 2021-07-08:
1. If you use MKS EXP-MOT3 on E3D, you need's use EXP3 interface
2. You need set it. For example：
You need use X2 axis：
A. Configuration_adv.h file: enable  #define X_DUAL_STEPPER_DRIVERS
B. pins_MKS_ROBIN_E3_common.h add pin define（On M1）
```
#define X2_ENABLE_PIN  PC1//BEEPER
#define X2_STEP_PIN  PA4//LCD_EN
#define X2_DIR_PIN  PA5//LCD_RS
```
[image]
[image]

countcobolt, 2021-07-08:
Stupid Question, but where do I then set up the LCD display?

kind regards

mks-viva, 2021-07-08:
> Stupid Question, but where do I then set up the LCD display?
> 
> kind regards

If you must connect your lcd on e3 board, you can use MKS EXP-MOT3's M3 interface

countcobolt, 2021-07-08:
sorry to keep on asking stupid questions, but I only see EXP1 & EXP2 on the MOT3

countcobolt, 2021-07-15:
Hi all, do you mean : if you need to use exp3 you CAN NOT use the lcd or can? Sorry not clear to me

mks-viva, 2021-07-15:
If you use exp3, so, you only M3 interface is can be use

closed 2021-07-31 by mks-viva

tanjunior, 2024-09-07:
@mks-viva Please help me too. 

E3D UART mode
MOT-EXP UART mode
E3D EXP1 <-> EXP-MOT EXP1
E3D EXP2 <-> EXP-MOT EXP2

Following the [BTT EXP-MOT user guide](https://github.com/bigtreetech/BTT-Expansion-module/blob/master/BTT%20EXP-MOT/BTT%20EXP-MOT%20V1.0User%20Guide.pdf)

BTT-MOT EXP2 M1STP = E3D EXP2 PB15
BTT-MOT EXP2 M1DIR = E3D EXP2 PB0
BTT-MOT EXP2 M1EN = E3D EXP2 PC10
BTT-MOT EXP1 M1RX = E3D EXP1 PA7

```
[stepper_z1]
step_pin: EXP2_6
dir_pin: !EXP2_5
enable_pin: !EXP2_7
microsteps: 16
rotation_distance: 8
endstop_pin: ^PC6
endstop_pin: probe:z_virtual_endstop

[tmc2209 stepper_z1]
uart_pin: EXP1_6
run_current: 0.8
stealthchop_threshold: 999999

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PC1, EXP1_3=PA4, EXP1_5=PA6, EXP1_7=PC4, EXP1_9=<GND>,
    EXP1_2=PC3, EXP1_4=PA5, EXP1_6=PA7, EXP1_8=PC5, EXP1_10=<5V>,
    # EXP2 header
    EXP2_1=PB14, EXP2_3=PB11, EXP2_5=PB0,  EXP2_7=PC10,  EXP2_9=<GND>,
    EXP2_2=PB13, EXP2_4=PA15, EXP2_6=PB15, EXP2_8=<RST>, EXP2_10=<NC>,
    # CR_EXP3 header - additional header dedicated to Creality Ender 3/5 stock screens
    CR_EXP3_10=<5V>, CR_EXP3_8=PA4, CR_EXP3_6=PA6, CR_EXP3_4=<RST>, CR_EXP3_2=PC3,
    CR_EXP3_9=<GND>, CR_EXP3_7=PA5, CR_EXP3_5=PB0, CR_EXP3_3=PB11,  CR_EXP3_1=PC1
```

`FORCE_MOVE STEPPER=stepper_z1 DISTANCE=10 VELOCITY=5`
[image]