makerbase-mks/MKS-Robin-E3-E3D #122: In Marlin 2.0.9.1, 250000 baud rate printer stops randomly. In Marlin 2.0.7 and before, the baud rate of 250000 was running smoothly

TimeJammer, 2021-07-11:
Hello to everyone. I have mks E3D card and mks tft 35. In Marlin 2.0.9.1, 250000 baud rate printer stops randomly. In Marlin 2.0.7 and before, the baud rate of 250000 was running smoothly for 1.5 years.
In version 2.0.9.1 it is stable only at 115200. What is the reason of this ?

I did not cancel the original screen of the printer (Ender 3) and I use it together with tft 35. //#define SERIAL_PORT_2 The -1 setting does not work in the new marlin and I disabled it because it was said to be disabled in the new suggestions. Could this be the reason? While other versions do not have this problem, what could be the reason why 250000 baud rate does not work stable only in 2.0.9.1. Could it be caused by something or a change that I did wrong in this version? Thanks for your responses.

Motherboard: Mks E3D(tmc2209)
Screen: Mks TFT35 with wireless.
Ank mks 3d touch sensor.

Thanks in advance for your returns.

mks-viva, 2021-07-15:
If the new version of marlin does not have any feature updates, it is recommended that you use the original version directly.

TimeJammer, 2021-07-15:
All right, thanks. It's okay to stay on Marlin 2.0.7, right?

TimeJammer, 2021-07-15:
> If the new version of marlin does not have any feature updates, it is recommended that you use the original version directly.

All right, thanks. It's okay to stay on Marlin 2.0.7, right?

mks-viva, 2021-07-15:
Yes

TimeJammer, 2021-07-15:
Thanks for the answer. Have a nice day.

closed 2021-07-15 by TimeJammer