makerbase-mks/MKS-Robin-E3-E3D #125: robin e3d  with EXP-MOT3,  how to support three Extruders？

woaikuancheng0, 2021-07-23:
I want to support 3 extruders.
does those code right

//add e1
#define E1_STEP_PIN                         PA4
#define E1_DIR_PIN                          PA5
#define E1_ENABLE_PIN                       PC1
//add e2
#define E2_STEP_PIN                         PA6
#define E2_DIR_PIN                          PA7
#define E2_ENABLE_PIN                       PC1
//add e3
#define E3_STEP_PIN                         PC4
#define E3_DIR_PIN                          PC5
#define E3_ENABLE_PIN                       PC1

mks-viva, 2021-07-26:
Yes

closed 2021-07-31 by mks-viva

madurani, 2021-08-21:
What if I want to use uart mode(HW Robin Nano v2)? How will it be with CS and EN port, since they are on the same pin on scheme.
For example when I need to add second Z2-driver on M1, then I add in pin file:
```
#define Z2_ENABLE_PIN                        PE13
#define Z2_STEP_PIN                          PD13
#define Z2_DIR_PIN                           PC6
#ifndef Z2_CS_PIN
  #define Z2_CS_PIN                          PE13
#endif
```
How will it be with display? Options **#define TFT_LVGL_UI** and **#define TOUCH_SCREEN?** Should be commented in config file?
Or is there any other possibility to use display  MKS TS35?

tanjunior, 2024-08-20:
@woaikuancheng0 i am also planning to expand my board for a z2 axis to do tramming. how do you plug in the expansion board?