makerbase-mks/MKS-Robin-E3-E3D #126: FYSETC-Mini-12864 v2.1 or BTT Mini 12864 v10

kjanek, 2021-07-24:
Is it possible to make this screen work on the MKS Robin E3D? I already fixed some problems like:
* marlin not building correctly with the screen enabled in config
* neopixel not working for the screen
But except that the Backlight works and that I can change the background colour it doesn't show any letters. It even doesn't show the marlin logo on bootup.
Any way to fix this?

mks-viva, 2021-07-26:
Hi! You can refer to here #127

kjanek, 2021-07-26:
> Hi! You can refer to here #127
I really appreciate your help. This solution worked for me. I enabled 2 things in the marlin config and the screen (BTT Mini 12864 v1.0) is working perfectly fine now.
The enabled stuff:
#define MKS_MINI_12864_V3
and
#define NEOPIXEL_LED

closed 2021-07-26 by kjanek

mks-viva, 2021-07-27:
It's will work?

kjanek, 2021-07-27:
> It's will work?

Yes it works.

mks-viva, 2021-07-27:
> > It's will work?
> 
> Yes it works.

Download the latest marlin？Can you provide your firmware to me?

kjanek, 2021-07-27:
> > > It's will work?
> > 
> > 
> > Yes it works.
> 
> Download the latest marlin？Can you provide your firmware to me?

Yes of course I can provide my firmware to you.
I dont use the latest marlin. I use the marlin bugfix 2.0.x from 20210723.
For making the screen work i just enabled
#define MKS_MINI_12864_V3
and
#define NEOPIXEL_LED
in the marlin and reverted the connector for exp1 and exp2 in the board.
Send you the firmware in atachment and a photo of working screen.
[Marlin.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/6883476/Marlin.zip)
[mks_robin_e3.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/6883454/mks_robin_e3.zip)
[image]

mks-viva, 2021-07-27:
> > > > It's will work?
> > > 
> > > 
> > > Yes it works.
> > 
> > 
> > Download the latest marlin？Can you provide your firmware to me?
> 
> Yes of course I can provide my firmware to you.
> I dont use the latest marlin. I use the marlin bugfix 2.0.x from 20210723.
> For making the screen work i just enabled
> #define MKS_MINI_12864_V3
> and
> #define NEOPIXEL_LED
> in the marlin and reverted the connector for exp1 and exp2 (I turned them by 180 degrees) on the board.
> Send you the firmware in atachment and a photo of working screen.
> And here is the whole marlin folder: https://t.me/MKS_Robin_E3D_Anet_ET4_Plus/5
> [Marlin.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/6883476/Marlin.zip)
> [mks_robin_e3.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/6883454/mks_robin_e3.zip)
> [image]

Looks good😁

kjanek, 2021-07-27:
> Looks good😁

I know 😃 . So thanks again for your help.
I have one question to you.
Could you add a subsection in here https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki
about the FYSETC Mini 12864 v2.1, BTT Mini 12864 v1.0 so that other don't have to search everywhere for the the solution how to get this panel to working.
You could write something like this:
"For enabling the FYSETC Mini 12864 v2.1 or the BTT Mini 12864 v1.0 you have to do following steps:
1. Enable the MKS MINI 12864 V3 screen in Marlin Configuration.h (delete // before this line):
_#define MKS_MINI_12864_V3_
2. Enable the Neopixel led in Marlin Configuration.h (delete // before this line):
_#define NEOPIXEL_LED_
3. Compile and flash the firmware
4. Connect the EXP1 and EXP2 cable to the screen and to the motherboard if it still doesn't work this means you have to rotate the plug 180 degrees."
This would be a nice instruction to follow. Can you add it there?

mks-viva, 2021-07-27:
> > > > > > It's will work?
> > > > > 
> > > > > 
> > > > > Yes it works.
> > > > 
> > > > 
> > > > Download the latest marlin？Can you provide your firmware to me?
> > > 
> > > 
> > > Yes of course I can provide my firmware to you.
> > > I dont use the latest marlin. I use the marlin bugfix 2.0.x from 20210723.
> > > For making the screen work i just enabled
> > > #define MKS_MINI_12864_V3
> > > and
> > > #define NEOPIXEL_LED
> > > in the marlin and reverted the connector for exp1 and exp2 (I turned them by 180 degrees) on the board.
> > > Send you the firmware in atachment and a photo of working screen.
> > > And here is the whole marlin folder: https://t.me/MKS_Robin_E3D_Anet_ET4_Plus/5
> > > [Marlin.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/6883476/Marlin.zip)
> > > [mks_robin_e3.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/6883454/mks_robin_e3.zip)
> > > [image]
> > 
> > 
> > Looks goodgrin
> 
> I know  . So thanks again for your help.
> I have one question to you.
> Could you add a subsection in here https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki
> about the FYSETC Mini 12864 v2.1, BTT Mini 12864 v1.0 so that other don't have to search everywhere for the the solution how to get this panel to working.
> You could write something like this:
> "For enabling the FYSETC Mini 12864 v2.1 or the BTT Mini 12864 v1.0 you have to do following steps:
> 
> 1. Enable the MKS MINI 12864 V3 screen in Marlin Configuration.h (delete // before this line):
>    _#define MKS_MINI_12864_V3_
> 2. Enable the Neopixel led in Marlin Configuration.h (delete // before this line):
>    _#define NEOPIXEL_LED_
> 3. Compile and flash the firmware
> 4. Connect the EXP1 and EXP2 cable to the screen and to the motherboard if it still doesn't work this means you have to rotate the plug 180 degrees."
>    This would be a nice instruction to follow. Can you add it there?

I am happy to do this.

kjanek, 2021-07-27:
> I am happy to do this. https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/FYSETC_BTT_MINI12864

Thanks.
I have one more question to you. I am using my MKS Robin E3D with TMC2209. I set them to UART mode. But the Z stepper motor does strange noise when it's in UART mode. It's much quiter when its in dir mode. Now is there any way to make it quite in the UART mode. Or is the any way that it doesnt show TMC connection error when i change the mode for the Z to DIR mode?

mks-viva, 2021-07-27:
You only disable ** #define Z_DRIVER_TYPE  TMC2209** and use jumpers connect to M0,M1 (16 microstep)