makerbase-mks/MKS-Robin-E3-E3D #127: BTT mini12864 compile fail，

zhongxb, 2021-07-24:
Hi! You need add it in ini file
custom_marlin.NEOPIXEL_LED = Adafruit NeoPixel=https://github.com/ellensp/Adafruit_NeoPixel
[image]

_Originally posted by @mks-viva in https://github.com/makerbase-mks/MKS-Robin-E3-E3D/issues/114#issuecomment-862281285_

can't find response with https://github.com/ellensp/Adafruit_NeoPixel

404 not found

mks-viva, 2021-07-26:
You can refer to here https://github.com/MarlinFirmware/Marlin/pull/22368, the latest version marlinhas supported it

mks-viva, 2021-07-27:
https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/FYSETC_BTT_MINI12864

closed 2021-07-31 by mks-viva