makerbase-mks/MKS-Robin-E3-E3D #129: 3 fans on robin e3

NilsHarald, 2021-08-06:
Can someone point me to any document on how to setup 3 fans?
Hotendfan, partfan and boardfan. If there is a way to control them.

pauloeduardogodoy, 2021-08-06:
I am really interested in this type of setup too!

mks-viva, 2021-08-07:
@NilsHarald @pauloeduardogodoy 
Only add fan1,fan2 pin define in pins_MKS_ROBIN_E3_common.h file and connect external MOS module
Add pin define
```
#define FAN1_PIN   PC4  //Open: M106 P1 S255, Close: M107
#define FAN2_PIN   PC5  //Open: M106 P2 S255, Close: M107
```

MOS module: https://www.aliexpress.com/item/32824688674.html

pauloeduardogodoy, 2021-08-07:
Why MOS modules are necessary to connect simple fans? This approach looks a little bit overkill...

mks-viva, 2021-08-07:
Because FAN signal is control N channel MOS, so, need MOS modules

closed 2021-08-09 by mks-viva

bevanweiss, 2021-09-18:
@pauloeduardogodoy it's because the Robin E3/E3d only have a single output connector configured with MOSFET for PWM control of fan  So if you want more than 1 fan controlled by PWM, then you need to take one of the digital low current IO pins, configure it for output, and then feed that to a MOSFET so that you don't burn up the output pin.  The output pins are not capable of supplying more than a few mA, this is enough to drive a FET, but definitely not enough to drive a fan.  Hence you need something like the MOS module.  You could just use your own custom PCB with a small SOT-23 FET on it...

If you were using a 5V fan, then you could use the 5V rail from the three pin plugs, like the neopixel, then switch the FET with the digital output.  The adapter board could be quite 'neat' then.
If you're using a 12V / 24V supply for the fan, you'd need to reference the GND/0V of it back to the Robin E3/E3d GND also.