makerbase-mks/MKS-Robin-E3-E3D #130: Como carregar o Marlin para a placa Mks Robin e3d

Emeale2021, 2021-08-07:
Olá, sou novo aqui e se estiver fazendo algo errado me avisem.
Estou trocando a placa da minha impressora 3d caseira por uma MKS Robin e3d e gostaria de uma orientação se possível, atualmente uso a placa Arduino mega que configurava pela IDE do arduino, mas ainda não sei instalar o Marlin na MKS, alguém poderia me dar uma dica?

pauloeduardogodoy, 2021-08-07:
Boa tarde @Emeale2021 seja bem vindo. Não tenho muita experiência ainda com essa placa, tanto é que estourei duas ligando step down XL4015 nelas... Queimei o transistor Q3 que controla a saída dos FANs 1 e 2...
De qualquer maneira, compilar o Marlin 2.0.x pra ela é bem simples. Só precisa instalar o Visual Basic Code e duas extensões nele. Sica essas instruções:
 - [PlatformIO](https://marlinfw.org/docs/basics/install_platformio_vscode.html)
 - [Auto Build Marlin](https://marlinfw.org/docs/basics/auto_build_marlin.html)

Depois faça as configurações necessárias nos arquivos "Configuration.h" e "Configuration_adv.h"

mks-viva, 2021-08-09:
You need use Vscode comlipe marlin for e3d board.

mks-viva, 2021-08-09:
https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/Marlin_firmware

closed 2021-08-09 by mks-viva