makerbase-mks/MKS-Robin-E3-E3D #132: Cannot compile Marlin 2.0.9.1

pauloeduardogodoy, 2021-08-16:
I am trying to compile the latest Marlin without success. Even using the Makerbase configuration website. Is there a step-by-step procedure to be followed? Thanks.

pauloeduardogodoy, 2021-08-17:
### It is freezes here:

PS C:\Marlin> platformio run -e mks_robin_e3
Processing mks_robin_e3 (platform: ststm32@~12.1; board: genericSTM32F103RC; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 (12.1.1) > STM32F103RC (48k RAM. 256k Flash)        
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (stlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.10900.200819 (1.9.0)
 - framework-cmsis 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TMCStepper> 0.7.1      
|   |-- <SoftwareSerial> 1.0
|   |-- <SPI> 1.0
|-- <U8glib-HAL> 0.5.0      
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <EEPROM> 2.0.1
|-- <SoftwareSerial> 1.0
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio\build\mks_robin_e3\FrameworkArduinoVariant\PeripheralPins.c.o
Compiling .pio\build\mks_robin_e3\FrameworkArduinoVariant\variant.cpp.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\Print.cpp.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\RingBuffer.cpp.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\USBSerial.cpp.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\VirtIOSerial.cpp.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\WInterrupts.cpp.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\WSerial.cpp.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\WString.cpp.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\abi.cpp.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\avr\dtostrf.c.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\board.c.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\hooks.c.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\itoa.c.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\main.cpp.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\new.cpp.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\pins_arduino.c.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\stm32\OpenAMP\libmetal\device.c.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\stm32\OpenAMP\libmetal\generic\condition.c.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\stm32\OpenAMP\libmetal\generic\cortexm\sys.c.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\stm32\OpenAMP\libmetal\generic\generic_device.c.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\stm32\OpenAMP\libmetal\generic\generic_init.c.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\stm32\OpenAMP\libmetal\generic\generic_io.c.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\stm32\OpenAMP\libmetal\generic\generic_shmem.c.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\stm32\OpenAMP\libmetal\generic\time.c.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\stm32\OpenAMP\libmetal\init.c.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\stm32\OpenAMP\libmetal\io.c.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\stm32\OpenAMP\libmetal\log.c.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\stm32\OpenAMP\libmetal\shmem.c.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\stm32\OpenAMP\mbox_ipcc.c.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\stm32\OpenAMP\open-amp\remoteproc\remoteproc_virtio.c.o
Compiling .pio\build\mks_robin_e3\FrameworkArduino\stm32\OpenAMP\open-amp\rpmsg\rpmsg.c.o
[... 384 lines cut ...]
Archiving .pio\build\mks_robin_e3\lib06f\libTMCStepper.a
Compiling .pio\build\mks_robin_e3\libd86\U8glib-HAL\clib\u8g_com_arduino_hw_spi.c.o
Compiling .pio\build\mks_robin_e3\libd86\U8glib-HAL\clib\u8g_com_arduino_hw_usart_spi.c.o
Compiling .pio\build\mks_robin_e3\libd86\U8glib-HAL\clib\u8g_com_arduino_no_en_parallel.c.o
Compiling .pio\build\mks_robin_e3\libd86\U8glib-HAL\clib\u8g_com_arduino_parallel.c.o
Archiving .pio\build\mks_robin_e3\lib469\libWire.a
Compiling .pio\build\mks_robin_e3\libd86\U8glib-HAL\clib\u8g_com_arduino_port_d_wr.c.o
Compiling .pio\build\mks_robin_e3\libd86\U8glib-HAL\clib\u8g_com_arduino_ssd_i2c.c.o
Compiling .pio\build\mks_robin_e3\libd86\U8glib-HAL\clib\u8g_com_arduino_st7920_custom.c.o
Compiling .pio\build\mks_robin_e3\libd86\U8glib-HAL\clib\u8g_com_arduino_st7920_hw_spi.c.o
Compiling .pio\build\mks_robin_e3\libd86\U8glib-HAL\clib\u8g_com_arduino_st7920_spi.c.o
Compiling .pio\build\mks_robin_e3\libd86\U8glib-HAL\clib\u8g_com_arduino_std_sw_spi.c.o
Compiling .pio\build\mks_robin_e3\libd86\U8glib-HAL\clib\u8g_com_arduino_sw_spi.c.o
[... cut ...]
Compiling .pio\build\mks_robin_e3\libd86\U8glib-HAL\clib\u8g_com_atmega_st7920_spi.c.o
Compiling .pio\build\mks_robin_e3\libd86\U8glib-HAL\clib\u8g_com_atmega_sw_spi.c.o
Compiling .pio\build\mks_robin_e3\libd86\U8glib-HAL\clib\u8g_com_atxmega_hw_spi.c.o
Compiling .pio\build\mks_robin_e3\libd86\U8glib-HAL\clib\u8g_com_atxmega_st7920_hw_spi.c.o
Compiling .pio\build\mks_robin_e3\libd86\U8glib-HAL\clib\u8g_com_esp32_ssd_i2c_fn.cpp.o
Compiling .pio\build\mks_robin_e3\libd86\U8glib-HAL\clib\u8g_com_i2c.c.o
Compiling .pio\build\mks_robin_e3\libd86\U8glib-HAL\clib\u8g_com_io.cpp.o

mks-viva, 2021-08-17:
Please provide your source code to me and i try to compile it.

pauloeduardogodoy, 2021-08-17:
I got it done! Do not know how I did it, but it is working now!

The only thing I did was get the files I created using the MKSTOOL again, made one change and compile, another change, and compile again, and so on... After 20+ compilations, I got everything set and compiled.

I am closing this thread. Thank you!

closed 2021-08-17 by pauloeduardogodoy