makerbase-mks/MKS-Robin-E3-E3D #133: Z axis not homing on delta firmware

happy-zelebob, 2021-08-18:
There is a problem with the MKS Robin E3D v1.1 board, presumably. I put the usual Ender 3 firmware on it, everything worked fine. Next, I installed modified firmware of the delta printer (Anycubic Kossel), after which when I send the head to the home point (Autohome), the Z axis is not sent to the endstop switch. If in detail, then first the X-axis is sent, everything is fine with it, the Y-axis, everything is fine with it, and then the Z-axis is lowered by some distance. In this case, if I press "Move Z", then all motors are working and the head is evenly move down vertically. I checked the operation of the motor and limit switch, everything is fine with them, I checked whether I indicated the correct board in the firmware, this is also okay (v 1.1). What could be the problem?

My configuration file 
[configuration.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/7009383/configuration.zip)

mks-viva, 2021-08-19:
The board use to Ender3 and it's will work, so, no problem with the hardware. I think you need check you endstop settings

happy-zelebob, 2021-08-20:
It was realy problem with endstops. I needed to use z min plug, not z max.

closed 2021-08-20 by happy-zelebob