makerbase-mks/MKS-Robin-E3-E3D #134: Wanhao i3 plus

131710cc, 2021-08-18:
Can i use this board to upgrade my wanhao i3 plus?

if so, there is any tutorial that can help me do it. i can only find ender 3 tutorials

Is the drv8825 compatible to this board?

mks-viva, 2021-08-19:
This is a 4axis board, if you select E3D board, you can select MKS TMC2209 or other drivers, if you select E3 board and it's integrated 2209 driver uart mode.

131710cc, 2021-08-19:
> This is a 4axis board, if you select E3D board, you can select MKS TMC2209 or other drivers, if you select E3 board and it's integrated 2209 driver uart mode.

but there is a "how to" to upgrade my wanhao?

mks-viva, 2021-08-20:
Melzi picture provide to me.

mks-viva, 2021-08-21:
You only use 2 cable connect to E3D board