makerbase-mks/MKS-Robin-E3-E3D #135: REPRAP_DISCOUNT_SMART_CONTROLLER

131710cc, 2021-08-31:
[image]
This is how is my lcd after connect to my e3d, with a sd or not, it do not show any character

mks-viva, 2021-09-01:
Please provide your source code to me.

mks-viva, 2021-09-07:
I has test it and it's will work. May be you need check your wiring and your lcd.
[Robin_e3.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/7119646/Robin_e3.zip)
[image]

131710cc, 2021-09-07:
Could you send me the correct way to wiring this lcd? cause it looks right in the picture.

mks-viva, 2021-09-13:
Connect the EXP1 and EXP2 cable to the screen and to the motherboard if it still doesn't work this means you have to rotate the plug 180 degrees.
[image]

closed 2021-09-23 by mks-viva