makerbase-mks/MKS-Robin-E3-E3D #136: MKS robin e3d v1.1+TMC2130 SPI Fail

Warober, 2021-09-10:
Hi I have a problem with MKS robin e3d v1.1.
The problem is that if I load Marlin (2.0.9.1) with the TMC2130, the MKS MINI12864 V3 LCD screen freezes at startup, the Marlin logo appears and crashes. I have tried compiling with TMC2130_STANDALONE and it boots perfectly. I have also tried another screen and the error is the same.
The bug is caused by activating the SPI function of the drivers.
Thanks greetings

mks-viva, 2021-09-13:
https://github.com/MarlinFirmware/Marlin/issues/22755

mks-viva, 2021-09-16:
You need use the latest version marlin test

mks-viva, 2021-09-16:
My test is ok, you can download test it
[Marlin-bugfix-2.0.x-E3D-TMCSPI.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/7174366/Marlin-bugfix-2.0.x-E3D-TMCSPI.zip)
[Robin_e3.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/7174367/Robin_e3.zip)

closed 2021-09-23 by mks-viva