makerbase-mks/MKS-Robin-E3-E3D #137: stepper motor move only in one direction

zeck7, 2021-09-12:
Hi, this my first time upgrade to a new board. i had a problem with my X motor only move in left direction After some troubles shoot I've swap with other the stepper driver, swap connector with Y motor, change connector pin(move to opposite direction but only that one direction) and change inverted in fimware and It seems that pin on board of X motor always move to one direction. im using mks robin e3d and tmc2209.

mks-viva, 2021-09-13:
According to your description, it should be caused by the 2209 module

zeck7, 2021-09-13:
i've also have swap with different stepper driver

zeck7, 2021-09-13:
is there a way I could remap the dir pin (pb2) in the board?

mks-viva, 2021-09-14:
Maybe it is caused by the firmware, you try to return to zero first, and then move the X axis.

zeck7, 2021-09-14:
ive done that too. still moving one direction. Is there a way i could jumper X dir signal from boaaard to stepper driver?

mks-viva, 2021-09-14:
Can you provide pictures?

zeck7, 2021-09-16:
[image]
I found the silkscreen marking but I don't know how solder? No contact pad available.

zeck7, 2021-09-16:
[image]
And I just noticed on schematic "X_dir" not wiring directly like "Z_DIR", "E0_DIR", and "Y_DIR" pin? Is this the main cause that my stepper X motor move only one direction?

bevanweiss, 2021-09-18:
X_DIR has a pulldown resistor to ensure it has a valid state at boot up.  This is very unlikely to be your problem.
So you have tried with a different stepper driver in this position?
Which version of Marlin are you running?  How did you source this version, and if you compiled it yourself, what configs do you have setup? (attach your Configuration.h and Configuration_adv.h)

Are you using the UART mode for the TMC2209, or stand-alone?  What jumpers do you have configured under the TMC2209 for the X-axis stepper?

zeck7, 2021-09-18:
Im using 
marlin2.0
tmc_2099, uart mode.
jumper at board as uart mode with one jumper.

At first I'm using "Marlin-2.0.6.1_for_ender3_TMC2209.bin" file from makerbase precompile .bin files at here https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/master/firmware/V1.1/Firmware%20bin%20for%20ender3 . Yup I've change file name to "Robin_e3.bin" first and copy to root directory of sd card and  it boot like normal.  Every tool on my ender3 work like it should be they're are normal except my X axis motor which only move in one direction.
Then...
I thought maybe I need to change "INVERT_X_DIR" from "false'' to ''true"  in firmware so I tried compile marlin 2.0 and Mks e3d v1.1 config file from "Marlin example config files". And after I change what recommended, I copy the bin file to sd card and screen boot like normal then I check each axis, tools, endstop, they function as intended except at X axis always move only in one direction.

One more things:
On ender 3 screen  if I move X axis +10mm, my X axis move to the left.
and then if I move back X axis -10mm, my X axis still move to the left

bevanweiss, 2021-09-18:
You should double check the following file:
Marlin\src\pins\stm32f1\pins_MKS_ROBIN_E3_common.h
To confirm that X_DIR_PIN is defined as PB2

You could get a multimeter and check if the X_DIR pin (at R58, or on the step stick) is actually changing.
You might have an issue with the resistor array (R58)

zeck7, 2021-09-18:
I have multimeter but I don't know how use / measure. Can you guide me how? as how to setting the multimeter reading at V, or Ohm or continuity and where -ve probe to touch and +ve probe to touch. Thank you so much 🙏🏻🙏🏻

bevanweiss, 2021-09-18:
Have your stepper free from any mechanical constraints, definitely don't have it driving a physical axis.

Have the firmware without the direction inversion, and then make it do a really long move in the positive direction.  then measure the points shown (you can measure on the top of the step stick).
[image]
Then do a really long move in the reverse direction, and again measure the points shown.

Then change the firmware to have direction inversion, and do the same two tests.

If you don't get any voltage at all (you'd be expecting to see about 3.3V for a '1' and ~0V for a '0'), then that suggests that you've got a board problem.  Possibly firmware, possibly uC, possibly traces from uC pin to resistor array, possibly resistor array, possibly traces from resistor array to step stick socket pin.
[image]
[image]

If you do get changes in voltage, then it's your stepper driver that's not working.

To repair the uC or traces would be quite advanced.  So at that stage you'd be in a better situation to just replace the board.  Unless you have a friend that is good with fine pitch electronics.  You could get them to re-solder the PB2 pin... it's possibly a dry solder joint and is not making contact.  But that's incredibly unlikely..  and it would be easy to make things worse with solder bridges.

zeck7, 2021-09-18:
thanks for guiding me.
after measure the voltage across  "dir" and "gnd" 
I got 2.9v  when I increased/move the value  of X axis "the plus  direction" (motor spinning CW)
and 
I got 0.2v when I decreased/move the value of X axis "the minus direction" (yet motor still spin CW)

and one more thing I observe the voltage stay constant at 2.9v when I move motor to plus(+) direction even the motor finish it rotation.
And same goes to if I move the motor to minus(-) direction the voltage reading stay constant at 0.2v even after the motor has done it rotation.

zeck7, 2021-09-18:
additional info.
I tried your experiment to read the voltage across  Y and Z axis just for sake of comparing and I notice there is massive different in voltage reading.
at Y and Z axis if I move the axis to plus direction I got 3.6v
and I got much near to zero volt or at 0.02v if I reverse the direction
compare to my problem X axis high at 2.6V and low at 0.2V

zeck7, 2021-09-18:
God please give this man a shit load reward.
bevanweiss you are the best!! I found out that if I move the "driver IC power" jumer pin to 3.3v my  X axis problem gone!.
 I dont know why eventho im not plan on using sensorless homing.

zeck7, 2021-09-18:
I LOVE YOU SO MUCH!
bevanweiss you just restore the faith in humanity. Most people said "Buy new board" but you keep digging with me and teching me new thing on how to use multimeter and finding the possible problem. God bless you 👼🏻👼🏻

bevanweiss, 2021-09-18:
It might be worth you removing the step stick and either measuring from the bottom of the main board, or by putting some pin headers into the step stick sockets to measure again.
The Drive IC voltage select of 3.3V / 5V is separate from the uC, so should not have changed the DIR pin voltage.
That it did have an impact on the DIR pin voltage makes me think that it's your step-stick... and perhaps it's not 5V tolerant.

You do have it working now though, so not much to be gained from further investigation.
You should probably close this issue, I'd be pointing the finger at your step sticks (and the use of 5V supply voltage for them).

closed 2021-09-21 by zeck7

zeck7, 2021-09-21:
apologize Im so exited with this new board. So far my printer all working like charm I've install BL_touch and BMG extruder recently and print all calibration test i could possible find on internet and all setting dial in marvelously. 
Do you means "step-stick" is "STEP pin"? so i should measure "STEP" pin and "GRD" pin like before you told me to measure "DIR" and "GRD". Means with multimeter "black probe" at "GND pin" and "red probe" at "STEP pin" right?

bevanweiss, 2021-09-21:
No, the stepstick is the driver board.  So my suggestion was to measure the DIR pin without having the driver board installed.  To see if the lower voltage was caused by the mainboard just presenting a low voltage, or by something on the driver pulling the voltage low.

But since it's all working, no need to dig deeper.