makerbase-mks/MKS-Robin-E3-E3D #138: mks robin e3d v1.1 How to upload bootloader?

ShWoong, 2021-09-19:
Almost Stm32F103 boards are having bootmode pin(or bootmode switch) but I can't found that pin in my board.
Plz help me... In my case I connected PC with MKS Board with enclosed blue usb cable to Serial Modul.
but STM Cube programmer could not find Board. but I confirmed connected COM5.

I want upload bootloader for Arduino in my board. Plz help me.
Thank you

bevanweiss, 2021-09-21:
BOOT0 and BOOT1 are on the chip
[image]

They are tied to 0V (for BOOT0) and pulled down to 0V (for BOOT1).  Hence boot occurs using pre-programmed FLASH.
This will likely be a custom MKS bootloader.
If you want to overwrite this bootloader, then you will need to use SWD via J1.

STM Cube wouldn't see the ST bootloader because it only uses that if the FLASH is empty, or BOOTx pins are configured to force ROM programmed bootloader use.
You could desolder the BOOT0 pin, lift the leg, tack on a wire and drive it, and apply a signal to BOOT1 too.. but if you just want Arduino, then you'd better find a different board.
You could probably shoe-horn Arduino into a binary image and then use the MKS bootloader to get it into the board.. but that seems like more trouble than it's worth.  I'd go the SWD way instead.

mks-viva, 2021-09-22:
You can use ST-link upload it and need not change hardware.

ShWoong, 2021-09-23:
Thank you your support. That means Do not change STM32 chip and pin setting right?

mks-viva, 2021-09-23:
Yes, you need not change hardware.

pauloeduardogodoy, 2021-09-24:
I have this board installed into my Ender-5 Pro + 4 TCM2209 drivers and configured with a custom Marlin bug-fix.

I would like just to flash the bootloader to remove the beep sound and the "boot" wording in the start.
Ia the an easy and straightforward guide/instructions explaining how to proceed?

Thanks!

bevanweiss, 2021-09-24:
Use ST-link or other similar SWD debugger tool
https://www.st.com/en/development-tools/st-link-v2.html

and then burn the firmware from here
https://github.com/makerbase-mks/MKS-Robin-E3-E3D/tree/master/firmware/Bootloader

J1 is the connector for the SWD interface.

mks-viva, 2021-09-24:
[image]

closed 2021-09-24 by mks-viva