makerbase-mks/MKS-Robin-E3-E3D #139: STM32F103 / STM32F407 boards should use on-chip USB

ag88, 2021-10-04:
hi MKS,
Your boards are quite good, but STM32F103 / STM32F407 boards should use on-chip USB support instead of using a separate USB-UART 

There is a problem, as seen in this PR for Marlin
https://github.com/MarlinFirmware/Marlin/pull/22529
Because STM32F103 and STM32F407 UART only has a single byte buffer. When the microcontroller is busy it can skip bytes from the host. So it becomes necessary to increase interrupt priority for the UART, this comes at a cost to interrupts to other peripherals such as HardwareTimer.

Both STM32F103 and STM32F407 has on-chip USB support, please revise your designs to use the on-chip USB rather than using a separate USB-UART bridge.

e.g. your MKS-Nano-V3 is correctly designed
https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X
https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/blob/main/hardware/MKS%20Robin%20Nano%20V3.0_004/MKS%20Robin%20Nano%20V3.0_004%20SCH.pdf
pins PA11 USB D- and PA12 USB D+ goes direct to USB.
But please update your designs so that the Nano-V1 and Robin E3 etc
https://github.com/makerbase-mks/MKS-Robin-E3-E3D
https://github.com/makerbase-mks/MKS-Robin-E3-E3D/blob/master/hardware/MKS%20Robin%20E3%20V1.1_004/MKS%20Robin%20E3%20V1.1_004%20SCH.pdf
use pins PA11 USB D- and PA12 USB D+ goes direct to USB.

ag88, 2021-10-07:
Lets update the 'old' product to 'new' versions, because those boards are popular, maybe call it Nano V1.1U usb E3 vx.yU / E3D x.yU etc. using USB direct helps to solve some 3d printing problems on STM32 boards, especially with Marlin.

mks-viva, 2021-10-07:
Yes, good proposal