makerbase-mks/MKS-Robin-E3-E3D #142: Save color or set default color with Mks 12864 mini

acca84, 2021-10-04:
Hello,
I have a Mks 12864 mini used with robin E3D motherboard.
It works very well, i can change the color on the lcd using the LCD menu but can't save the default color.
Is it possible to change the default color or save the desired one ?
thanks

reefwarrior, 2022-01-23:
> Hello, I have a Mks 12864 mini used with robin E3D motherboard. It works very well, i can change the color on the lcd using the LCD menu but can't save the default color. Is it possible to change the default color or save the desired one ? thanks

Hi,

How did you get the screen to work? And which firmware did you use? 

Tinus

acca84, 2022-01-24:
Hi, i'm using Marlin 2.0 and using : 
#define MKS_MINI_12864_V3