makerbase-mks/MKS-Robin-E3-E3D #147: MKS E3D v1 with TMC2209 & pancake stepper

Ne0nx3r0, 2021-11-18:
I'm having an issue where there's not nearly enough torque using the ender 3 with the speedrive mod and a pancake stepper.

My setup is:

* MKS E3D v1 
* TMC2209 (in UART)
* 42BYGH 23MM 1.5A Pancake stepper 

I've tried adjusting the current in the firmware to 500, 700 and 850, also tried disabling stealthchop, but the motor simply doesn't seem to have enough torque.

I have the same setup on a MKS E3D v1.1 board that is working fine, so I'm confused why this one is not. 

Someone mentioned that using a TMC2208 driver might resolve this issue as the v1 board can be buggy with 2209 drivers, is that a possibility?

Ne0nx3r0, 2021-11-20:
For what it's worth, here's the firmware I'm using:
https://github.com/Ne0nx3r0/robin_e3d_TMC2209_bltouch_safe_v1.1_speeddrive/blob/main/Marlin/Configuration_adv.h

https://github.com/Ne0nx3r0/robin_e3d_TMC2209_bltouch_safe_v1.1_speeddrive/blob/main/Marlin/Configuration.h

mks-viva, 2021-11-23:
MKS Robin E3D V1.0 and V1.1 compare：
[image]

closed 2021-11-30 by mks-viva