makerbase-mks/MKS-Robin-E3-E3D #148: MKS-Robin-E3D_V1.1 FAN PA8 issues

TheHorris, 2021-11-23:
Hello,

Setup:
- MKS E3D v1
- TMC2209 V 2.0 (in UART)
- DC 12V Part cooling fan

i have some issues to run my Part cooling fan with the board on the "PA 8 Pin"
The fan works fine with the 12V output but when I try to control it with the PA 8 pin with 100% nothing happens. Im working with the Marlin 2.0 bugfix firmware.
I see no issues with the board.

thanks for help.

Horris

Firmware:
https://github.com/TheHorris/MKS-Robin-E3D-V1.1_Marlin-firmware/blob/main/Configuration.h

https://github.com/TheHorris/MKS-Robin-E3D-V1.1_Marlin-firmware/blob/main/Configuration_adv.h

pauloeduardogodoy, 2021-11-23:
Are you using a 12v PSU powering all system or just using a step-down module to reduce from 24v to 12v?

TheHorris, 2021-11-23:
I use a 12V PSU

- S-250-12 12V 20A DC power supply

mks-viva, 2021-11-30:
FAN can be control by M106 S255?

TheHorris, 2021-11-30:
sure, the display says 100% but the fan on the PIN PA8 dont run.

mks-viva, 2021-12-03:
In other words, there is no voltage output on the FAN interface?

TheHorris, 2021-12-05:
correct. There is no measurable voltage on this output.

mks-viva, 2021-12-13:
If you control FAN by command, the FAN Mos led will on?
[image]

TheHorris, 2021-12-13:
There is an LED?! :D No, by command the FAN Mos led is off.

mks-viva, 2021-12-23:
If you send M106 and D11 led will on, if you send M107 and it's will off. So, it means that this interface is fine