makerbase-mks/MKS-Robin-E3-E3D #161: Firmware update stopped on "update 100%"

happy-zelebob, 2022-02-15:
I know about this problem https://github.com/makerbase-mks/MKS-Robin-E3-E3D/issues/101, I installed the right file, Robin_e3.bin from .pio/build/mks_robin_e3, but I still have this problem with both symptoms. 

My source code:
[Marlin-2.0.6.1.zip](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/files/8072734/Marlin-2.0.6.1.zip)

Seelenkind, 2022-02-19:
Is this your board ? 
[image]
No ? so change it

happy-zelebob, 2022-02-21:
This is my board, mks robin e3d v1.1

happy-zelebob, 2022-03-04:
I tried to compile the firmware on another computer, I tried to change SD card, no result. Any help for me, Makerbase?!