makerbase-mks/MKS-Robin-E3-E3D #164: PID autotune doesn't stop heating the bed.

pFallenAngelq, 2022-03-05:
I've installed a mains heater for the bed, have changed the values for the new thermistor (11 as it's an Keenovo thermal pad) in  configuration.h as well as thermal_protection_bed_period and watch_bed_temp_period in configuration_adv.h but the bed just keeps on raising the temp.
Started PID autotune set to 70°, reaches 70°, keeps 70° at first cycle, starts second cycle and pushed reset at 92° (went to make a coffee).

MKS-hosiey, 2022-03-11:
Will the heating bed automatically heat up when it is turned on again?

pFallenAngelq, 2022-03-18:
> Will the heating bed automatically heat up when it is turned on again?

It heats up when I set it to heat and stays at the requested temp, but can't do PID autotune because it keeps on heating up at second cycle.