makerbase-mks/MKS-Robin-E3-E3D #165: Extruder not working when MKS 12864 Mini V3 enabled

bschenke2, 2022-03-08:
I recently upgraded my board to a MKS Robin E3D V1.1 and had everything working fine with the factory Anet LCD Screen running Marlin bugfix_2.0.x. I decided to upgrade the LCD to a Makerbase MKS Mini 12864 V3. When I enable the features in the code to work the new screen, everything works fine except the extruder will no longer work. When I flash it back to the previous working firmware it works perfectly again. Below are the things I enabled for the new screen:

Enable: #define MKS_MINI_12864_V3
Enable: #define NEOPIXEL_LED

I also disabled the old screen when I added the new one.

Any ideas on where the problem may lie?

MKS-hosiey, 2022-03-16:
Does your new firmware board define V1.1?

guitosi, 2022-06-01:
Boa tarde conseguiu resolver o problema, estou com o mesmo e nao consigo atualizar a firmware com esse MKS_MINI_12864_V3

bschenke2, 2022-06-09:
I have still not been able to resolve this. I'm currently still using my original LCD controller from my printer.

Vit220, 2022-06-21:
I have this problem too
if disable SDSUPORT extruder work
but i need work SDSUPORT too
marlin 2.1 bugfix

guitosi, 2022-06-21:
Atualmente estou usando octoprint para enviar arquivos já que o SDsuport precisa estar desabilitado, mas eu gostaria de poder usar o SD também.

bschenke2, 2022-06-21:
Does the SD card still work from the main board if this is disabled, or does it disable both the SD slot in the LCD board as well as the main board?

almuneeb, 2022-06-21:
I have same issue with my board

bschenke2, 2022-06-22:
I just tested mine with the SDSUPPORT disabled, and I'm still able to use the card slot on the main board. It kind of defeats the purpose of having the SD slot on the controller, but at least I'm able to print and use the new controller. I use OctoPrint for 95% of my prints anyways, so it's not a huge deal to me I guess.

guitosi, 2022-06-22:
Acabei de fazer o teste, e na minha com sdsuporte desabilitado não consigo usar, pois nem aparece a opção no display, ou eu não estou sabendo configurar.