makerbase-mks/MKS-Robin-E3-E3D #169: Cooling Fan 100% all the time

kjanek, 2022-07-05:
The cooling fan shorted out on my E3D v1.1 most probably the cooling fan MOSFET most probably dead caus it's working all the time at 100%. Can I change it or somehow fix the fan? Thanks

pauloeduardogodoy, 2022-07-06:
If you tried to install a crapy step down to use a 12V fan like I did, yes you killed the SMD transistor.

It is possible to replace, but you need to find the correct part to buy and need an SMD soldering station... Almost impossible to do it with a common soldering iron.

kjanek, 2022-07-06:
> If you tried to install a crapy step down to use a 12V fan like I did, yes you killed the SMD transistor.
> 
> It is possible to replace, but you need to find the correct part to buy and need an SMD soldering station... Almost impossible to do it with a basic soldering iron.

I have not added a crappy step down. I had a 24V fan that had a little crappy connector and short cable. And throw one print the connector cables were pulled from connector and they shorted out. But still i think it made the same result as your issues. So basically the board needs a exchange :(

CrazyMark493, 2023-03-11:
i got the same problem
it was the sot-23 package mosfet died
buy some AO3400 mosfet with“A08K”on it and replace it

CrazyMark493, 2023-03-11:
> 

i did it with only a crappy soldering iron

Sylwester79, 2024-02-20:
Isn't it connected to the wrong output?