makerbase-mks/MKS-Robin-E3-E3D #174: This board support Core XY Printer

JungHoonLim, 2022-12-21:
Hello,

This board support Core XY Printer 
Can some body help me to configure the firmware for My 3D printer ?

i will use ender 3 Display

This is my printer 
[image]

pauloeduardogodoy, 2022-12-21:
This is a Marlin function... Uncomment #define COREXY line.
For more details read [this](https://all3dp.com/2/marlin-firmware-how-to-edit-marlin-firmware/) tutorial.

JungHoonLim, 2022-12-21:
> This is a Marlin function... Uncomment _#define COREXY_ line. For more details read [this](https://all3dp.com/2/marlin-firmware-how-to-edit-marlin-firmware/) tutorial.

thank U..^^